nbrachet/FlyDrones.iOS #13: Add menu to change mode

nbrachet, 2015-05-27:
When the user clicks on the current mode, a "bubble menu" should appear with the option to change the mode -- ie. a list of the following modes: STABILIZE, ALT_HOLD, LOITER, RTL, LAND, DRIFT, POSHOLD.
The current mode shouldn't be in the menu.

nbrachet, 2015-05-27:
Note: mode is changed by sending a SET_MODE message:
```
GCS: SEQ:182 SYSID:252 COMPID:  1 - SET_MODE: custom_mode=LOITER target_system=1 base_mode=CUSTOM_MODE
```

nab0y4enko, 2015-05-28:
Done in 0.1.4.2

nbrachet, 2015-06-02:
menu should close once a heartbeat message is received

nab0y4enko, 2015-06-09:
Fixed in version 0.1.5
I recorded a video with simulation of the process
https://youtu.be/A95kKF0Ak4Y

nbrachet, 2015-06-09:
Nope. Menus still stay displayed.

closed 2015-06-15 by nbrachet