nbrachet/FlyDrones.iOS #14: Incorrect heading displayed by the compass

nbrachet, 2015-05-27:
With the following MAVLink message:
SEQ:110 SYSID:  1 COMPID:  1 - VFR_HUD: airspeed=0.71 groundspeed=0.71 alt=15.34 climb=0.01 heading=153 throttle=0

Compass shows a heading of 0 (North).

nbrachet, 2015-06-02:
Broken again in 0.1.4.4

nab0y4enko, 2015-06-15:
It works correctly in v0.2

nab0y4enko, 2015-07-16:
Please provide tlog for testing

nab0y4enko, 2015-07-17:
in a file 2015-07-16 15-03-27.tlog no information about MAVLINK_MSG_ID_VFR_HUD.
In previous logs this information has been.

nbrachet, 2015-07-17:
MAVLINK_MSG_ID_VFR_HUD is sent if the app requests it via a REQUEST_DATA_STREAM/EXTRA2:
```
GCS: SEQ:  4 SYSID:252 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=2 target_system=1 target_component=0 req_stream_id=EXTRA2 START
```

This message should be sent every time the connection is established.

Technically, REQUEST_DATA_STREAM/EXTENDED_STATUS should also be sent to receive GPS_RAW_INT messages:
```
GCS: SEQ: 14 SYSID:252 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=2 target_system=1 target_component=0 req_stream_id=EXTENDED_STATUS START
```

See [GCS_Mavlink.cpp](https://github.com/diydrones/ardupilot/blob/af8b9e703cffa4be6670168ca9d1a623ac4b54da/ArduCopter/GCS_Mavlink.cpp#L731) for reference.

nab0y4enko, 2015-07-17:
Done in 0.2.5.5

nbrachet, 2015-07-22:
I don't see it:
```
/dev/ttyAMA0 opened
Logging to /var/log/tlogs/2015-07-22 11-48-37.tlog
Connecting to 108.26.177.27:5555
Connected to 108.26.177.27:5555 via 192.168.2.159:59797
GCS connection
UAS sysid=1
UAS: SEQ:159 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
UAS is in STANDBY
UAS control mode STABILIZE
UAS is DISARMED
UAS: SYS_STATUS: onboard_control_sensors_present=(3D_GYRO|3D_ACCEL|3D_MAG|ABSOLUTE_PRESSURE|GPS|ANGULAR_RATE_CONTROL|ATTITUDE_STABILIZATION|YAW_POSITION|Z_ALTITUDE_CONTROL|XY_POSITION_CONTROL|MOTOR_OUTPUTS|AHRS|TERRAIN) onboard_control_sensors_enabled=!(Z_ALTITUDE_CONTROL&XY_POSITION_CONTROL) onboard_control_sensors_health=(*all*) load=2,611.2% voltage_battery=11.403V current_battery=0.39A drop_rate_comm=0.00% errors_comm=0 errors_count1=0 errors_count2=0 errors_count3=0 errors_count4=0 battery_remaining=99%
UAS: SEQ:170 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
UAS: SEQ:183 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
UAS: SEQ:197 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
UAS: SEQ:209 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
   : loadavg: 0.10 0.21 0.21 1/73 3475
UAS: SEQ:222 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
GCS: SEQ: 32 SYSID:252 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
UAS: SEQ:234 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
GCS: SEQ: 33 SYSID:252 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
UAS: SEQ:246 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
GCS: SEQ: 34 SYSID:252 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
UAS: SEQ:  2 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
GCS: SEQ: 35 SYSID:252 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
```

nbrachet, 2015-07-27:
The REQUEST_DATA_STREAM message is not sent after the heartbeat message is lost.