nbrachet/FlyDrones.iOS #15: No MAVLink message sent when arming/disarming

nbrachet, 2015-05-27:
In 0.1.4.1, no MAVLink message is sent to arm/disarm.

The following message is expected to arm:
```
GCS: SEQ:  1 SYSID:252 COMPID:  1 - COMMAND_LONG: COMPONENT_ARM_DISARM: param1=1 param2=0 param3=0 param4=0 param5=0 param6=0 param7=0 confirmation=1
```
and:
```
GCS: SEQ:  1 SYSID:252 COMPID:  1 - COMMAND_LONG: COMPONENT_ARM_DISARM: param1=0 param2=0 param3=0 param4=0 param5=0 param6=0 param7=0 confirmation=1
```
to disarm

nab0y4enko, 2015-05-28:
Done in 0.1.4.2

nbrachet, 2015-06-01:
target_system is incorrect:
```
GCS: SEQ:158 SYSID:255 COMPID:  0 - COMMAND_LONG: COMPONENT_ARM_DISARM: target_sys=0 target_comp=1 ARM confirmation=1
```
should be:
```
GCS: SEQ:158 SYSID:255 COMPID:  0 - COMMAND_LONG: COMPONENT_ARM_DISARM: target_sys=1 target_comp=1 ARM confirmation=1
```
Assuming of course the UAS' system id is 1.

nbrachet, 2015-06-02:
Fixed in 0.1.4.4

closed 2015-06-02 by nbrachet