nbrachet/FlyDrones.iOS #16: No RC_OVERRIDE message are sent

nbrachet, 2015-05-27:
Only HEARTBEAT messages are sent to the UAS.

nab0y4enko, 2015-05-28:
> Only RC_OVERRIDE messages should be sent.
> Also they should only be sent when the UAS is ARMED.

I added a check:
```
if (!(currentStatus.mavBaseMode & (uint8_t)MAV_MODE_FLAG_SAFETY_ARMED)) {
        return nil;
}
```
Therefore it is not sent

nbrachet, 2015-05-28:
But the UAS was armed...

nab0y4enko, 2015-05-28:
I'll check!

nab0y4enko, 2015-05-28:
In version 0.1.4.3 override_rc message are sent when the status ARMED. And vice versa

nbrachet, 2015-05-28:
What do you mean by "vice versa"? They are not sent when the status is
DISARMED?

nab0y4enko, 2015-05-28:
Sorry. Do not sent. :)

nbrachet, 2015-06-01:
Double check that the target_system is set to the UAS's system id in the MAVLINK_MSG_ID_RC_CHANNELS_OVERRIDE message...

nbrachet, 2015-06-02:
Confirmed in 0.1.4.4: no RC_OVERRIDE message are sent when the UAS is armed,

nab0y4enko, 2015-06-02:
application sends messages:
tag 11 - heartbeat
tag 12 - MAVLINK_MSG_ID_RC_CHANNELS_OVERRIDE
[image]

nbrachet, 2015-06-02:
Not sent to the UAS:
```
GCS: SEQ: 99 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:100 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:101 SYSID:  1 COMPID:  0 - COMMAND_LONG: COMPONENT_ARM_DISARM: target_sys=1 target_comp=0 ARM confirmation=1
GCS: SEQ:102 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:103 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:104 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:105 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:106 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:107 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:108 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:109 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:110 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:111 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:112 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:113 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:114 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:115 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:116 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:117 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:118 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:119 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:120 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
GCS: SEQ:121 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
```

nab0y4enko, 2015-06-02:
Nick, can you give me your utility to check?

nbrachet, 2015-06-02:
Those are the logs from the UAS.

You can simulate that with something like:
```
socat -lf xyz -v TCP-LISTEN:5555,reuseaddr TCP-LISTEN:5556,reuseaddr
mavplayback --out=tcp:127.0.0.1:5555 ...
```
Then you can decode the log file (xyz) with ```mavinspector```

nab0y4enko, 2015-06-02:
Nick, I did not succeed checked using socat. But I checked with a simple script.
heartbeat once per second (tag 11)
RC_OVERRIDE sent 10 times per second (tag 12)
server receives the data correctly

App log:
```
2015-06-02 20:44:34.465 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:34.565 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:34.665 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:34.765 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:34.865 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:34.965 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:35.065 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:35.165 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:35.265 FlyDrones[2515:205525] Did send data for tag:11
2015-06-02 20:44:35.266 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:35.365 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:35.466 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:35.565 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:35.666 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:35.766 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:35.865 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:35.965 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:36.066 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:36.165 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:36.264 FlyDrones[2515:205525] Did send data for tag:11
2015-06-02 20:44:36.266 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:36.366 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:36.466 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:36.566 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:36.665 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:36.765 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:36.865 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:36.965 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:37.065 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:37.165 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:37.265 FlyDrones[2515:205483] Did send data for tag:11
2015-06-02 20:44:37.265 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:37.365 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:37.465 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:37.565 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:37.665 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:37.765 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:37.865 FlyDrones[2515:205490] Did send data for tag:12
2015-06-02 20:44:37.965 FlyDrones[2515:205490] Did send data for tag:12
2015-06-02 20:44:38.065 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:38.165 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:38.264 FlyDrones[2515:205483] Did send data for tag:11
2015-06-02 20:44:38.265 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:38.366 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:38.466 FlyDrones[2515:205483] Did send data for tag:12
2015-06-02 20:44:38.564 FlyDrones[2515:205488] Did send data for tag:12
2015-06-02 20:44:38.665 FlyDrones[2515:205488] Did send data for tag:12
2015-06-02 20:44:38.765 FlyDrones[2515:205488] Did send data for tag:12
2015-06-02 20:44:38.865 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:38.965 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:39.065 FlyDrones[2515:205525] Did send data for tag:12
2015-06-02 20:44:39.166 FlyDrones[2515:205525] Did send data for tag:12
```
This log from console:
```
2015-06-02 20:44:34.426221 received data: ?
F????kg
2015-06-02 20:44:34.526337 received data: ?
                                           F??????
2015-06-02 20:44:34.626569 received data: ?
                                           F?????K
F????'?-02 20:44:34.726387 received data: ?
2015-06-02 20:44:34.826418 received data: ?F????mP
[... 28 lines cut ...]
2015-06-02 20:44:37.226156 received data: ?)F????E
2015-06-02 20:44:37.328407 received data: ?*F????J?
2015-06-02 20:44:37.426308 received data: ?+F?????^
2015-06-02 20:44:37.527922 received data: ?,F??????
2015-06-02 20:44:37.626319 received data: ?-F????r
2015-06-02 20:44:37.733366 received data: ?.F????L?
2015-06-02 20:44:37.826190 received data: ?/F?????i
2015-06-02 20:44:37.926962 received data: ?0F?????~
2015-06-02 20:44:38.025164 received data: ?	???
2015-06-02 20:44:38.028066 received data: ?2F????^e
2015-06-02 20:44:38.126735 received data: ?3F??????
2015-06-02 20:44:38.226199 received data: ?4F?????I
2015-06-02 20:44:38.325836 received data: ?5F?????
2015-06-02 20:44:38.426186 received data: ?6F????XR
2015-06-02 20:44:38.525496 received data: ?7F??????
2015-06-02 20:44:38.626458 received data: ?8F?????
2015-06-02 20:44:38.726282 received data: ?9F?????
2015-06-02 20:44:38.825598 received data: ?:F????R

2015-06-02 20:44:38.926409 received data: ?;F??????
2015-06-02 20:44:39.025725 received data: ?	???
2015-06-02 20:44:39.027926 received data: ?=F?????
```

[image]

nbrachet, 2015-06-02:
Humm... They seem to be showing up now:
```
GCS: SEQ: 59 SYSID:  1 COMPID:  0 - RC_CHANNELS_OVERRIDE: chan1_raw=1,515 chan2_raw=1,528 chan3_raw=1,563 chan4_raw=1,500 chan5_raw=0 chan6_raw=0 chan7_raw=0 chan8_raw=0 target_system=1 target_component=0
```
However the SYSID is wrong: set to 1.

nab0y4enko, 2015-06-09:
Default sysid all set to 1

nbrachet, 2015-06-29:
Again no RC_OVERRIDE messages, despite:
```
UAS: SEQ: 23 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT|ARMED system_status=CRITICAL mavlink_version=3
```

nbrachet, 2015-07-06:
Example of RC MAP not being download upon connection:
```
(gdb) run  -v -h264 -h 720p25 -b 2m newton
Starting program: /home/nick/userland.git/build/bin/mavraspivid -v -h264 -h 720p25 -b 2m newton
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb6a2a460 (LWP 4279)]
[New Thread 0xb622a460 (LWP 4280)]
[New Thread 0xb5a2a460 (LWP 4281)]
[New Thread 0xb522a460 (LWP 4282)]

mavraspivid Camera App v1.3.12-nick (f2b56c0-debug)

Width 1280, Height 720 (720p)
bitrate 2.00MBits/s, framerate 25, intraperiod 25 (1s)
H264 Profile high (low-latency)
H264 Quantisation level 0, Inline headers
Sending to 108.26.177.27:5555 (RTP/UDP pt=96)
Master /dev/ttyAMA0 @ 57600 baud to 108.26.177.27:5555 (TCP)

Preview No
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Sensor mode = 0, Focus mode = fixed-infinity

MAVLink PIXHAWK 1.0 (Sat May 30 00:50:08 2015)

/dev/ttyAMA0 opened
Logging to /var/log/tlogs/2015-07-06 15-59-08.tlog
Connecting to 108.26.177.27:5555
Connected to 108.26.177.27:5555 via 192.168.2.172:53804
GCS connection
UAS: SEQ:104 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
UAS sysid=1
   : SEQ:  0 SYSID:255 COMPID:  1 - COMMAND_LONG: REQUEST_AUTOPILOT_CAPABILITIES: target_sys=1 target_comp=0 param1=1 param2=0 param3=0 param4=0 param5=0 param6=0 param7=0 confirmation=0
   : SEQ:  1 SYSID:255 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=1 target_system=1 target_component=0 req_stream_id=RAW_SENSORS START
   : SEQ:  2 SYSID:255 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=1 target_system=1 target_component=0 req_stream_id=EXTENDED_STATUS START
   : SEQ:  3 SYSID:255 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=1 target_system=1 target_component=0 req_stream_id=RC_CHANNELS START
UAS is in STANDBY
UAS control mode STABILIZE
UAS is DISARMED
UAS: SEQ:105 SYSID:  1 COMPID:  1 - COMMAND_ACK: REQUEST_AUTOPILOT_CAPABILITIES: UNSUPPORTED
UAS: SEQ:106 SYSID:  1 COMPID:  1 - RAW_IMU: time_usec=91,428,261 xacc=21 yacc=-37 zacc=-952 xgyro=0 ygyro=0 zgyro=0 xmag=-25 ymag=199 zmag=397
UAS: SEQ:107 SYSID:  1 COMPID:  1 - SCALED_IMU2: time_boot_ms=93,981 xacc=17 yacc=-18 zacc=-975 xgyro=-3 ygyro=-11 zgyro=-8 xmag=17 ymag=234 zmag=381
UAS: SEQ:108 SYSID:  1 COMPID:  1 - SCALED_PRESSURE: time_boot_ms=93,981 press_abs=1,015.02 press_diff=-0.00671875 temperature=4,291
UAS: SEQ:109 SYSID:  1 COMPID:  1 - SYS_STATUS: onboard_control_sensors_present=(3D_GYRO|3D_ACCEL|3D_MAG|ABSOLUTE_PRESSURE|GPS|ANGULAR_RATE_CONTROL|ATTITUDE_STABILIZATION|YAW_POSITION|Z_ALTITUDE_CONTROL|XY_POSITION_CONTROL|MOTOR_OUTPUTS|AHRS|TERRAIN) onboard_control_sensors_enabled=!(Z_ALTITUDE_CONTROL&XY_POSITION_CONTROL) onboard_control_sensors_health=(*all*) load=15.7% voltage_battery=12.150V current_battery=0.67A drop_rate_comm=0.00% errors_comm=0 errors_count1=0 errors_count2=0 errors_count3=0 errors_count4=0 battery_remaining=99%
UAS: SYS_STATUS: onboard_control_sensors_present=(3D_GYRO|3D_ACCEL|3D_MAG|ABSOLUTE_PRESSURE|GPS|ANGULAR_RATE_CONTROL|ATTITUDE_STABILIZATION|YAW_POSITION|Z_ALTITUDE_CONTROL|XY_POSITION_CONTROL|MOTOR_OUTPUTS|AHRS|TERRAIN) onboard_control_sensors_enabled=!(Z_ALTITUDE_CONTROL&XY_POSITION_CONTROL) onboard_control_sensors_health=(*all*) load=15.7% voltage_battery=12.150V current_battery=0.67A drop_rate_comm=0.00% errors_comm=0 errors_count1=0 errors_count2=0 errors_count3=0 errors_count4=0 battery_remaining=99%
UAS: SEQ:110 SYSID:  1 COMPID:  1 - POWER_STATUS: Vcc=5,219 Vservo=55 flags=1
UAS: SEQ:112 SYSID:  1 COMPID:  1 - MISSION_CURRENT: seq=0
UAS: SEQ:113 SYSID:  1 COMPID:  1 - GPS_RAW_INT: time_usec=0 lat=423,005,210 lon=-712,351,051 alt=46,350 eph=9,999 epv=65,535 vel=0 cog=0 fix_type=1 satellites_visible=0
UAS: SEQ:114 SYSID:  1 COMPID:  1 - NAV_CONTROLLER_OUTPUT: nav_roll=0 nav_pitch=0 alt_error=-0.0685107 aspd_error=0 xtrack_error=0 nav_bearing=250 target_bearing=0 wp_dist=0
UAS: SEQ:116 SYSID:  1 COMPID:  1 - SERVO_OUTPUT_RAW: time_usec=91,468,285 servo1_raw=1,005 servo2_raw=1,005 servo3_raw=1,005 servo4_raw=1,005 servo5_raw=1,005 servo6_raw=1,005 servo7_raw=0 servo8_raw=0 port=0
UAS: SEQ:117 SYSID:  1 COMPID:  1 - RC_CHANNELS_RAW: time_boot_ms=94,021 chan1_raw=0 chan2_raw=0 chan3_raw=0 chan4_raw=0 chan5_raw=0 chan6_raw=0 chan7_raw=0 chan8_raw=0 port=0 rssi=0
GCS: SEQ: 75 SYSID:  1 COMPID:  0 - HEARTBEAT: custom_mode=0 type=GCS autopilot=INVALID base_mode=MANUAL_INPUT|ARMED system_status=ACTIVE mavlink_version=3
UAS: SEQ:118 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
UAS: SEQ:119 SYSID:  1 COMPID:  1 - RAW_IMU: time_usec=92,457,864 xacc=21 yacc=-37 zacc=-950 xgyro=0 ygyro=0 zgyro=-1 xmag=-24 ymag=198 zmag=396
UAS: SEQ:120 SYSID:  1 COMPID:  1 - SCALED_IMU2: time_boot_ms=95,011 xacc=24 yacc=-17 zacc=-978 xgyro=3 ygyro=-1 zgyro=2 xmag=13 ymag=235 zmag=380
UAS: SEQ:121 SYSID:  1 COMPID:  1 - SCALED_PRESSURE: time_boot_ms=95,011 press_abs=1,015 press_diff=-0.0238281 temperature=4,291
[... cut ...]
   : SEQ: 91 SYSID:255 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=1 target_system=1 target_component=0 req_stream_id=RAW_SENSORS START
   : SEQ: 92 SYSID:255 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=1 target_system=1 target_component=0 req_stream_id=EXTENDED_STATUS START
   : SEQ: 93 SYSID:255 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=1 target_system=1 target_component=0 req_stream_id=RC_CHANNELS START
[Thread 0xb522a460 (LWP 4282) exited]
[Thread 0xb5a2a460 (LWP 4281) exited]
[Thread 0xb622a460 (LWP 4280) exited]
[Thread 0xb6a2a460 (LWP 4279) exited]
[Inferior 1 (process 4278) exited normally]
(gdb) 
```

nbrachet, 2015-07-06:
Here's an example where PARAM_REQUEST_LIST is sent *after* the ARM command:
```
(gdb) run
Starting program: /home/nick/userland.git/build/bin/mavraspivid -v -h264 -h 720p25 -b 2m newton
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
[New Thread 0xb6a2a460 (LWP 4284)]
[New Thread 0xb622a460 (LWP 4285)]
[New Thread 0xb5a2a460 (LWP 4286)]
[New Thread 0xb522a460 (LWP 4287)]

mavraspivid Camera App v1.3.12-nick (f2b56c0-debug)

Width 1280, Height 720 (720p)
bitrate 2.00MBits/s, framerate 25, intraperiod 25 (1s)
H264 Profile high (low-latency)
H264 Quantisation level 0, Inline headers
Sending to 108.26.177.27:5555 (RTP/UDP pt=96)
Master /dev/ttyAMA0 @ 57600 baud to 108.26.177.27:5555 (TCP)

Preview No
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
Sensor mode = 0, Focus mode = fixed-infinity

MAVLink PIXHAWK 1.0 (Sat May 30 00:50:08 2015)

/dev/ttyAMA0 opened
Logging to /var/log/tlogs/2015-07-06 16-01-05.tlog
Connecting to 108.26.177.27:5555
Connected to 108.26.177.27:5555 via 192.168.2.172:53805
GCS connection
UAS: SEQ: 46 SYSID:  1 COMPID:  1 - RAW_IMU: time_usec=207,993,342 xacc=20 yacc=-37 zacc=-952 xgyro=0 ygyro=0 zgyro=-1 xmag=-24 ymag=197 zmag=396
UAS sysid=1
   : SEQ:  0 SYSID:255 COMPID:  1 - COMMAND_LONG: REQUEST_AUTOPILOT_CAPABILITIES: target_sys=1 target_comp=0 param1=1 param2=0 param3=0 param4=0 param5=0 param6=0 param7=0 confirmation=0
   : SEQ:  1 SYSID:255 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=1 target_system=1 target_component=0 req_stream_id=RAW_SENSORS START
   : SEQ:  2 SYSID:255 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=1 target_system=1 target_component=0 req_stream_id=EXTENDED_STATUS START
   : SEQ:  3 SYSID:255 COMPID:  1 - REQUEST_DATA_STREAM: req_message_rate=1 target_system=1 target_component=0 req_stream_id=RC_CHANNELS START
UAS: SEQ: 47 SYSID:  1 COMPID:  1 - SCALED_IMU2: time_boot_ms=210,546 xacc=22 yacc=-21 zacc=-981 xgyro=7 ygyro=-3 zgyro=5 xmag=12 ymag=233 zmag=379
UAS: SEQ: 48 SYSID:  1 COMPID:  1 - SCALED_PRESSURE: time_boot_ms=210,546 press_abs=1,015.04 press_diff=0.0160937 temperature=4,489
UAS: SEQ: 49 SYSID:  1 COMPID:  1 - COMMAND_ACK: REQUEST_AUTOPILOT_CAPABILITIES: UNSUPPORTED
UAS: SEQ: 50 SYSID:  1 COMPID:  1 - SYS_STATUS: onboard_control_sensors_present=(3D_GYRO|3D_ACCEL|3D_MAG|ABSOLUTE_PRESSURE|GPS|ANGULAR_RATE_CONTROL|ATTITUDE_STABILIZATION|YAW_POSITION|Z_ALTITUDE_CONTROL|XY_POSITION_CONTROL|MOTOR_OUTPUTS|AHRS|TERRAIN) onboard_control_sensors_enabled=!(Z_ALTITUDE_CONTROL&XY_POSITION_CONTROL) onboard_control_sensors_health=(*all*) load=17.0% voltage_battery=12.138V current_battery=0.70A drop_rate_comm=0.00% errors_comm=0 errors_count1=0 errors_count2=0 errors_count3=0 errors_count4=0 battery_remaining=99%
UAS: SYS_STATUS: onboard_control_sensors_present=(3D_GYRO|3D_ACCEL|3D_MAG|ABSOLUTE_PRESSURE|GPS|ANGULAR_RATE_CONTROL|ATTITUDE_STABILIZATION|YAW_POSITION|Z_ALTITUDE_CONTROL|XY_POSITION_CONTROL|MOTOR_OUTPUTS|AHRS|TERRAIN) onboard_control_sensors_enabled=!(Z_ALTITUDE_CONTROL&XY_POSITION_CONTROL) onboard_control_sensors_health=(*all*) load=17.0% voltage_battery=12.138V current_battery=0.70A drop_rate_comm=0.00% errors_comm=0 errors_count1=0 errors_count2=0 errors_count3=0 errors_count4=0 battery_remaining=99%
UAS: SEQ: 51 SYSID:  1 COMPID:  1 - POWER_STATUS: Vcc=5,230 Vservo=62 flags=1
UAS: SEQ: 53 SYSID:  1 COMPID:  1 - MISSION_CURRENT: seq=0
UAS: SEQ: 54 SYSID:  1 COMPID:  1 - GPS_RAW_INT: time_usec=0 lat=423,005,210 lon=-712,351,051 alt=46,350 eph=9,999 epv=65,535 vel=0 cog=0 fix_type=1 satellites_visible=0
UAS: SEQ: 55 SYSID:  1 COMPID:  1 - NAV_CONTROLLER_OUTPUT: nav_roll=0 nav_pitch=0 alt_error=0.0601709 aspd_error=0 xtrack_error=0 nav_bearing=250 target_bearing=0 wp_dist=0
UAS: SEQ: 57 SYSID:  1 COMPID:  1 - SERVO_OUTPUT_RAW: time_usec=208,073,094 servo1_raw=1,005 servo2_raw=1,005 servo3_raw=1,005 servo4_raw=1,005 servo5_raw=1,005 servo6_raw=1,005 servo7_raw=0 servo8_raw=0 port=0
UAS: SEQ: 58 SYSID:  1 COMPID:  1 - RC_CHANNELS_RAW: time_boot_ms=210,626 chan1_raw=0 chan2_raw=0 chan3_raw=0 chan4_raw=0 chan5_raw=0 chan6_raw=0 chan7_raw=0 chan8_raw=0 port=0 rssi=0
UAS: SEQ: 59 SYSID:  1 COMPID:  1 - HEARTBEAT: custom_mode=0 (STABILIZE) type=HEXAROTOR autopilot=ARDUPILOTMEGA base_mode=CUSTOM_MODE|STABILIZE|MANUAL_INPUT system_status=STANDBY mavlink_version=3
UAS is in STANDBY
UAS control mode STABILIZE
UAS is DISARMED
UAS: SEQ: 60 SYSID:  1 COMPID:  1 - RAW_IMU: time_usec=209,014,214 xacc=18 yacc=-38 zacc=-950 xgyro=-1 ygyro=0 zgyro=0 xmag=-23 ymag=196 zmag=396
UAS: SEQ: 61 SYSID:  1 COMPID:  1 - SCALED_IMU2: time_boot_ms=211,567 xacc=18 yacc=-18 zacc=-992 xgyro=8 ygyro=-9 zgyro=-1 xmag=16 ymag=233 zmag=375
UAS: SEQ: 62 SYSID:  1 COMPID:  1 - SCALED_PRESSURE: time_boot_ms=211,567 press_abs=1,015.03 press_diff=0.00359375 temperature=4,491
UAS: SEQ: 63 SYSID:  1 COMPID:  1 - SYS_STATUS: onboard_control_sensors_present=(3D_GYRO|3D_ACCEL|3D_MAG|ABSOLUTE_PRESSURE|GPS|ANGULAR_RATE_CONTROL|ATTITUDE_STABILIZATION|YAW_POSITION|Z_ALTITUDE_CONTROL|XY_POSITION_CONTROL|MOTOR_OUTPUTS|AHRS|TERRAIN) onboard_control_sensors_enabled=!(Z_ALTITUDE_CONTROL&XY_POSITION_CONTROL) onboard_control_sensors_health=(*all*)  [... cut ...]
UAS: SEQ:129 SYSID:  1 COMPID:  1 - PARAM_VALUE: param_value=950 param_count=417 param_index=108 param_id="RC10_MIN" param_type=4
UAS: SEQ:130 SYSID:  1 COMPID:  1 - PARAM_VALUE: param_value=1,500 param_count=417 param_index=109 param_id="RC10_TRIM" param_type=4
UAS: SEQ:131 SYSID:  1 COMPID:  1 - PARAM_VALUE: param_value=1,900 param_count=417 param_index=110 param_id="RC10_MAX" param_type=4
UAS: SEQ:132 SYSID:  1 COMPID:  1 - PARAM_VALUE: param_value=1 param_count=417 param_index=111 param_id="RC10_REV" param_type=2
UAS: SEQ:133 SYSID:  1 COMPID:  1 - PARAM_VALUE: param_value=0 param_count=417 par

closed 2015-07-14 by nbrachet