nbrachet/FlyDrones.iOS #17: Map window shows "Show Legend" and useless toggle

nbrachet, 2015-06-02:


nab0y4enko, 2015-06-09:
done in 0.1.5.1

nbrachet, 2015-06-09:
How can it show a "copter icon" on the map, and say N/A for the GPS fix?
[image]

nbrachet, 2015-06-09:
HDOP is expressed in 100m in the GPS_RAW_INT, so it needs to be divided by 100 before being displayed.

nab0y4enko, 2015-06-10:
i.e. always displayed?

nbrachet, 2015-06-10:
No. Just when displayed.

nab0y4enko, 2015-06-10:
You wrote that display GPS # sats and HDOP is necessary only when fix_type is> 2.
Then you wrote that display this information, Also if fix_type == 1
Do I understand correctly?

nbrachet, 2015-06-10:
Yes so when fix_type >= 1. But fix_type could still be 0 when the GPS fix is not available.

nab0y4enko, 2015-06-10:
If fix_type == 0, I should display the icon of helicopter?

nbrachet, 2015-06-10:
If fix_type == 0 there is no GPS fix, no GPS coordinates. Where would the icon be placed?

nab0y4enko, 2015-06-15:
Done in v0.2

nbrachet, 2015-07-17:
fix_type == 1 still indicates there is no GPS fix -- I misread that.
So when fix_type == 1 (and GPS coordinates are still available) just draw the map but with no icon.

nab0y4enko, 2015-07-17:
what text displayed if fixtype is 1?
"Satellites: N/A HDOP: N/A" ?

nbrachet, 2015-07-17:
I lost it now but I have seen examples where fix_type=1 and lat/long/hdop != NA. In that case I'd display the values in the message (and keep the map at that lat/lon).
If lat/lon/hdop is not in the message then yes: N/A.

nab0y4enko, 2015-07-17:
Done in 0.2.5.6

closed 2015-07-24 by nbrachet