nbrachet/FlyDrones.iOS #24: Image goes out and back without any video

nbrachet, 2015-06-15:
[image]

[image]

nab0y4enko, 2015-06-16:
Can you give more details?

nbrachet, 2015-06-16:
Not much more. I started playing the video then stopped it leaving the TCP connection alive. The screen shows the last image from the video (bottom screenshot). However once in a while the image would disappear and a black image would be displayed (top screenshot) for a few seconds, then the last image of the video would come back.

nab0y4enko, 2015-07-09:
fixed

closed 2015-07-15 by nbrachet