nbrachet/FlyDrones.iOS #25: Video playback is falling behind

nbrachet, 2015-06-18:
The video playback starts a few seconds behind (already high), and keeps falling behind further and further. I'm measured it roughly at over 1min behind after only a few minutes of playback.
The same feed played on a macosx laptop shows less than a second latency.

nab0y4enko, 2015-06-18:
Hi, Nick. Please make a test stream for me. I'll check and correct

nbrachet, 2015-06-19:
Here's a [5min](https://dl.dropboxusercontent.com/u/1020551/x.h264) video, with time overlay.

nab0y4enko, 2015-06-20:
hi Nick
I'm looking solution to the problem to the video delay
problem occurs during the decoding
ipad 3 is much worse to cope with it

model	h264cat		app
-------------------------------------------
ipad3 	50fps		33fps
		25fps		23fps

ipad4	50fps		42fps
		25fps		24fps

nbrachet, 2015-06-22:
How did you measure the app fps?

nab0y4enko, 2015-06-22:
Hi, Nick.
I have a tool that allows me to determine speed of rendering, in opengl
Now I am doing optimization of app.
I'm using ipad mini (first generation) for testing.
It has the same processor as the ipad2 (Apple a5)

nab0y4enko, 2015-06-23:
Nick, please, open for me the test stream

nbrachet, 2015-07-06:
Still happening, even at 480p 15fps. Got 30 seconds behind pretty quickly.

nab0y4enko, 2015-07-09:
fixed

nab0y4enko, 2015-07-16:
This task still relevant?

nbrachet, 2015-07-17:
Seems fixed.

closed 2015-07-17 by nbrachet