nbrachet/FlyDrones.iOS #32: Apparent loss of connection but no visual feedback

nbrachet, 2015-07-14:
[image]

nab0y4enko, 2015-07-15:
I need more details

nbrachet, 2015-07-15:
I don't have any unfortunately: the screen was apparently locked -- ie. no menus were working, neither was the map button -- yet the "waiting for heartbeat" message wasn't shown either.

nab0y4enko, 2015-07-15:
I can not repeat it, without access to the server

nab0y4enko, 2015-07-15:
I can not repeat it without access to the server

nbrachet, 2015-07-15:
Can you focus on #33 for now?

nab0y4enko, 2015-07-15:
https://github.com/nbrachet/FlyDrones.iOS/releases/tag/0.2.5.3 <https://github.com/nbrachet/FlyDrones.iOS/releases/tag/0.2.5.3>

nbrachet, 2015-07-15:
The first time the connection is lost the "waiting for heartbeat" message is shown, but not for the subsequent connection loss.

nab0y4enko, 2015-07-16:
Fixed in the next build

nbrachet, 2015-07-24:
Haven't seen it in a while. Closing.

closed 2015-07-24 by nbrachet