nbrachet/FlyDrones.iOS #34: Map window should stay up until the map icon is clicked

nbrachet, 2015-07-15:
Today the map window disappears as soon as the user clicks anywhere on the screen

nbrachet, 2015-07-15:
Same applies to the battery "menu"

nab0y4enko, 2015-07-16:
The iPad UI Element Guidelines section of the iPad’s Human Interface Guideline forbids displaying more than one popover element at a time. Let's come up with another version of the interface

nbrachet, 2015-07-16:
Does it mean the popup has to disappear when the user clicks anywhere on the screen?

nab0y4enko, 2015-07-16:
yes

nbrachet, 2015-07-17:
Still it shouldn't automatically close (even after a while)

nab0y4enko, 2015-07-17:
Map closes when server connection is broken or delay heartbeat
do not close?

nbrachet, 2015-07-17:
Not that's the intended behavior.
However sometimes it closes by itself, without the connection being broken.

nab0y4enko, 2015-07-17:
perhaps the connection is lost for a short time, and HUD is not shown

nbrachet, 2015-07-17:
Maybe... What is the timeout to decide the connection is down?

nab0y4enko, 2015-07-17:
Done in 0.2.5.6