nbrachet/FlyDrones.iOS #36: Add Altitude to the display

nbrachet, 2015-07-17:
Use a similar control as the one used to display heading, but vertical on the left to display altitude (from the VFR_HUD message.)

nab0y4enko, 2015-07-18:
What variant do you need?
[image]
[image]
[image]

nab0y4enko, 2015-07-18:
Template :)
[image]

nbrachet, 2015-07-19:
The first variant is closer to what I had in mind... But it would have to be transparent. I like the fading effect top/bottom but they should be more pronounced so roughly only 1/4 the height is readable.

I think it's just as well to use the same control as the one for the heading, but vertically on the left. Same size.

nab0y4enko, 2015-07-19:
Hi Nick. I understood you. I'll do it in the morning.

> On Jul 20, 2015, at 01:08, Nick Brachet <notifications@github.com> wrote:
> 
> The first variant is closer to what I had in mind... But it would have to be transparent. I like the fading effect top/bottom but they should be more pronounced so roughly only 1/4 the height is readable.
> 
> I think it's just as well to use the same control as the one for the heading, but vertically on the left. Same size.
>

closed 2015-07-22 by nbrachet